SOTAmat/SOTAcat #10: Memory leak

jeffkowalski, 2024-02-22:
Brian states:
Recent changes to the code seem to have caused a memory leak.  Actually I think it is a stack overflow since the ESP32 ESP-IDF framework does not have a dynamically adjusting stack and instead you have to declare your stack size for each thread.  Predicting what that stack size needs to be is hard and error prone.  The error manifests as a random reboot after sitting idle for a while my Windows PC browser has the sotacat.local page open (which refreshes the battery every minute).  In fact, the browser will call the battery voltage REST and the battery Percentage REST sometimes at the same moment, and I suspect the WiFi code gets an overflow.

jeffkowalski, 2024-02-22:
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/heap_debug.html#how-to-diagnose-memory-leaks will be useful here

brianmathews, 2024-02-23:
Letting the SOTACAT run on the bench while logging was enabled, I saw what I thought was our memory leak/crash/stack overflow reboot issue.

Turns out this is NOT a leak/crash causing the reboot (at least not what I saw).
The log below shows:

1. the unit's auto-sleep timer has reached the inactivity limit and it performs a shutdown to deep sleep sequence.
2. The unit appears to properly go to sleep
3. Somehow while in deep sleep, a WiFi event occurs and wakes the device.  I did not know this was even possible.  Everywhere I read in the docs there were two types of sleep: (a) light sleep where wifi might remain active, and (b) deep sleep where wifi is not active.
4. Upon waking, the unit calls a WiFi event handler which crashes because the sequence of entering deep sleep got rid of all the wifi context needed by the handler.
5. The crash of the handler causes the unit to do a full reboot
6. The reboot runs into the issue where (reported in another issue / bug ID) we think the ESP sends garbage to the KX radio causing the KX to lock up until rebooted.
7. The ESP32 keeps trying to connect to the (now hung) KX and loops forever until the 60 second boot watchdog shuts the unit down into deep sleep.
8. ...and here we go again!

Here is the log:

> I (3494178) SOTAmat: Blinks 4
I (3497778) SOTAmat: Battery voltage: 3.928 V
I (3497778) SOTAmat: Blinks 4
I (3501378) SOTAmat: Battery voltage: 3.930 V
I (3501378) SOTAmat: Blinks 5
I (3501828) SOTAmat: Powering off due to inactivity
I (3501828) SOTAmat: Entering deep sleep...
I (3501828) wifi:state: run -> init (0)
I (3501828) wifi:pm stop, total sleep time: 3079622303 us / 3500412134 us

I (3501828) wifi:<ba-del>idx
I (3501838) wifi:<ba-del>idx
I (3501838) wifi:new:<5,0>, old:<5,0>, ap:<255,255>, sta:<5,0>, prof:1
I (3501848) SOTAmat: WiFi event handler called with event_base: WIFI_EVENT, event_id: 5
I (3501858) SOTAmat: ============ Failed to connect to home WiFi, setting up stand-alone AP mode...
I (3501858) SOTAmat: Initializing WiFi in AP mode
ESP_ERROR_CHECK failed: esp_err_t 0x3014 (ESP_ERR_WIFI_STOP_STATE) at 0x4200324e
file: "src/wifi.c" line 33
func: wifi_init_ap
expression: esp_wifi_set_mode(WIFI_MODE_AP)

abort() was called at PC 0x403890a5 on core 0
Core  0 register dump:
MEPC    : 0x4038166a  RA      : 0x403890b0  SP      : 0x3fc9fec0  GP      : 0x3fc92e00
TP      : 0x3fc5e3c8  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000004  S1      : 0x3fc9ff24  A0      : 0x3fc9feec  A1      : 0x3fc9ff22
A2      : 0x00000000  A3      : 0x3fc9ff19  A4      : 0x00000001  A5      : 0x3fc99000
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fc92000  S3      : 0x3c0a148c
S4      : 0xffffffff  S5      : 0x00000000  S6      : 0xffffffff  S7      : 0x3fca05a0
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
MHARTID : 0x00000000

Stack memory:
3fc9fec0: 0x3c0a1e2c 0x42003252 0x3fc9ff20 0x4038fa92 0x00001800 0x00000018 0x3fca063c 0x3fc940f4
3fc9fee0: 0x3fc9ff24 0x3fc94110 0x3fc9ff20 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
3fc9ff00: 0x43502074 0x34783020 0x39383330 0x20356130 0x63206e6f 0x2065726f 0x00000030 0x00000000
3fc9ff20: 0x3fc90030 0x38333034 0x35613039 0x3c0a1400 0x3fc92000 0x3c0a148c 0x3fcacea4 0x403890a8
3fc9ff40: 0x3fc92000 0x3c0a148c 0x3fcacea4 0x42003252 0x4038c17c 0x4038c1a6 0x3fc9a11c 0x00000001
3fc9ff60: 0x00000003 0x41544f53 0x00746163 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9ff80: 0x00000000 0x34333231 0x38373635 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9ffa0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9ffc0: 0x00000000 0x00000107 0x00000004 0x00000400 0x00000000 0x00000000 0x00000000 0x00000000
3fc9ffe0: 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc92000 0x3c0a148c 0x3fc92000 0x42003344
3fca0000: 0xffffffff 0x3fc9f2f4 0x00000000 0x00000004 0x00000000 0x3fc9f2f4 0x3fca022c 0x4209cdf4
3fca0020: 0x00000000 0x3fc9f2f4 0x00000000 0x4209d538 0x00000101 0x3c0c6734 0x00000005 0x3fca8868
3fca0040: 0x00000101 0x3c0c6734 0x00000005 0x3fca8868 0x00000000 0x00000000 0x00000000 0x00000000
3fca0060: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc9f2f4 0x4209d66a
3fca0080: 0x00000000 0x00000000 0x00000000 0x4038c020 0x00000000 0x00000000 0x00000000 0x00000000
3fca00a0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca00c0: 0x00000154 0x3fc9fdd0 0x3fc99314 0x3fc95b64 0x3fc95b64 0x3fca00c4 0x3fc95b5c 0x00000005
3fca00e0: 0x3fca0668 0x3fca0668 0x3fca00c4 0x00000000 0x00000014 0x3fc9f5c0 0x5f737973 0x00747665
3fca0100: 0x1f8a245a 0x00dbae4a 0x00000000 0x3fca00b0 0x00000014 0x00000001 0x3fca8794 0x4203fea0
[... 114 lines cut ...]
I (1320) wifi:connected with whipitgood, aid = 33, channel 5, BW20, bssid = a0:36:bc:b6:e4:88
I (1330) wifi:security: WPA2-PSK, phy: bgn, rssi: -38
I (1340) wifi:pm start, type: 1

I (1340) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
I (1340) SOTAmat: WiFi event handler called with event_base: WIFI_EVENT, event_id: 4
I (1350) SOTAmat: ============ Success connecting to existing WiFi network...
I (1420) wifi:AP's beacon interval = 102400 us, DTIM period = 1
[... cut ...]
I (2360) esp_netif_handlers: sta ip: 192.168.50.51, mask: 255.255.255.0, gw: 192.168.50.1
I (2900) SOTAmat: No response received for baud rate 4800
I (3400) SOTAmat: No response received for baud rate 9600
I (3900) SOTAmat: No response received for baud rate 19200
I (4400) SOTAmat: No response received for baud rate 38400
I (4900) SOTAmat: No response received for baud rate 4800
I (5400) SOTAmat: No response received for baud rate 9600
I (5900) SOTAmat: No response received for baud rate 19200
I (6400) SOTAmat: No response received for baud rate 38400

jeffkowalski, 2024-02-27:
Perhaps we need to wait for WIFI_EVENT_[STA/AP]_STOP event after calling esp_wifi_stop(), to know it's really stopped?

brianmathews, 2024-02-27:
I don't know.  I have different ESP32 models in my prototypes that have very different behaviors in shutting down.  In fact some of the ESP-IDF shutdown routines crash on this XAIO ESP32C3.  The C3 is based on a different MCU with a different machine language and hardware than other ESP32's.
In other words, I am not an expert in shutdown.  I am surprised about the ability to wake as my understanding is that a wake is dependent on setting a trigger such as a clock/timer, or a GPIO pin transition, etc. and I haven't configured any triggers (unless there is a default?).

jeffkowalski, 2024-02-27:
I don't think it's waking.  I think that during the sleep process, various listeners are notified, including our wifi_event_handler, which may get a WIFI_EVENT_[STA/AP]_STOP.  The handler would be confused.
Just conjecturing. I'll dig in.

brianmathews, 2024-02-27:
You are correct.  The crash happens at shutdown, not later.  It just  looks that way in the sequence, but in reality the device isn't shut down.

The log shows timestamps in 1/1000ths of seconds (milliseconds):

- At 3501.828 we see the message "entering deep sleep"
- But there are many other messages, including the WiFi event handler, called at the same 3501.828 timestamp.  So we didn't really enter sleep and later crash.  We crash at about the same time as going to sleep.

jeffkowalski, 2024-02-28:
Interestingly, the bug doesn't seem to occur if the device is running in AP mode rather than STA mode.  If I cause it to fail to connect to my home Wifi, and allow it to broadcast an AP, then without connecting it to KX I see it correctly times out and goes into deep sleep:

```
...
I (60417) SOTAmat: Startup watchdog timer expired.  Shutting down.
I (60417) SOTAmat: Entering deep sleep...
I (60417) wifi:station: 5a:ce:f7:4b:3c:51 leave, AID = 1, bss_flags is 658547, bss:0x3fcaeb6c
I (60417) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (60427) wifi:<ba-del>idx:2, tid:0
I (60427) wifi:<ba-del>idx:3, tid:7
I (60437) SOTAmat: WiFi event handler called with event_base: WIFI_EVENT, event_id: 15
I (60447) SOTAmat: WiFi event handler called with event_base: WIFI_EVENT, event_id: 13
W (60457) httpd_txrx: httpd_sock_err: error in recv : 113
W (60457) httpd_txrx: httpd_sock_err: error in recv : 113
W (60467) httpd_txrx: httpd_sock_err: error in recv : 113
I (60487) wifi:flush txq
I (60487) wifi:stop sw txq
I (60487) wifi:lmac stop hw txq
I (60487) SOTAmat: Wifi stopped...
I (60487) SOTAmat: ADC deregister Curve Fitting calibration scheme
I (60487) SOTAmat: ADC shutdown...
I (60497) SOTAmat: GPIO's off and isolated...
I (60497) SOTAmat: Entering Deep Sleep...
I (60507) SOTAmat: Goodnight!
Disconnected (device reports readiness to read but returned no data (device disconnected or multiple access on port?))
Reconnecting to /dev/ttyACM0 ...........
```

closed 2024-02-28 by jeffkowalski

brianmathews, 2024-02-28:
Interesting.  I think in the error case the WiFi system we asked to shut down first does a network disconnect, which in turn might call our prior WiFi event handler, and might be trying to pass us a "disconnected from router" event.  We might need to disconnect our WiFi event handler after we connect (at boot time) or we may need it to accept a "disconnect" event which it isn't coded to handle.  Not sure.

commit 5b339ef1 (2024-02-28): fix #10: handle wifi disconn prior to deep sleep

When wifi was operating in station mode (e.g. connected to a home
network), radio timeout would cause and error in the wifi-handler

This error arises because of confusion in how to handle an incoming wifi
disconnect event.  We had been counting "disconnected" events to figure
out when to swap over to access-point mode, but during deep-sleep
shutdown, we of course don't want the wifi disconnection to do that.

This commit sifts the logic a bit to handle the two senses of
"disconnected" more finely.