SOTAmat/SOTAcat #11: SOTACAT and KX will stop talking to each other after a long period

jeffkowalski, 2024-02-22:
Brian states:
Another regent regression is that the SOTACAT and KX will stop talking to each other after a long period.  I think this happens after the SOTACAT reboots due to the prior stack overflow issue/crash because the Blue light comes on solid (meaning no connection to the radio during auto-baud-rate negotiation).  The SOTACAT will try forever to connect and fails to do so.  But something else is odd: if there is no connection to the radio after booting the device should go to deep sleep after 1 minute.  This sleep code works if you turn on your SOTACAT and lay it on your desk without a radio attached.  But after a crash reboot the behavior changes.  It might be that the light gets left on after a crash.  I don’t know.  Regardless, the only way to restore operation after this is to turn the radio off and back on again.   Unplugging SOTACAT and re-plugging doesn’t help.  So something is also happening on the radio side such that it refuses to listen to the serial port at some point.
Possible dup of #3?

jeffkowalski, 2024-02-22:
@brianmathews, would you kindly comment on whether this is a dup of #3 ?

brianmathews, 2024-02-22:
I don't know but I think this is actually 2 bugs:
1. A bug that causes a reboot.  Might be bug #10.
2. A but that causes boot message from UART to KX to cause KX to lock up (bug #2).

brianmathews, 2024-02-29:
Now that #10 has been fixed, are we still seeing this issue?

jeffkowalski, 2024-02-29:
I never experienced the issue, so I can't say for sure.  I originally opened this issue just to capture your initial comment. If you feel this is fixed, or perhaps simply vaguely duplicative of #10 and #2, then please close it.  We can always open a new issue with more concrete info and steps to reproduce.

brianmathews, 2024-02-29:
I do think this is a duplicate of #10 and/or #2 so I am closing this one.

closed 2024-02-29 by brianmathews