SOTAmat/SOTAcat #12: Web UI to read radio state

jeffkowalski, 2024-02-22:
Brian says:
I would like a web UI that allows me to at least “read” the state the radio is in (Frequency, Mode, power level, etc.).  And then the question is what would be a good UI for “write” and control of the radio from a cell phone beyond click-to-pounce?  For example: record / play CQ audio memories #1 and #2?  View transmit power level?  Toggle transmit power from maximum to 0.0 watts and back (for testing) so we don’t have to spin the dial forever?  Quick enable/disable of noise reduction?  A “smart Tune” that steps up or down by 3KHz and tunes, and then returns to the operating frequency so as to not “step” on conversations?  Etc.  Be creative.

jeffkowalski, 2024-06-30:
077d9dc4d9bee66b531b356bcb6103c612a804dd now implements some basic CAT control:
- Playback of audio memory.
- Transmission of arbitrary CW messages.
- Toggling transmit power min/max.

Not yet present: remote VFO and ATU

jeffkowalski, 2025-05-23:
VFO now present.
Do we need ATU?

poynting, 2025-05-23:
ATU would be nice.

See also #63 which AFAIK is still an active issue related to ATU not
switching when switching bands to spot.

jeffkowalski, 2025-05-23:
Thanks Justin.  Got it.  Now that we have an easy +/- 5K, it should be reasonable to have ATU.  

Workflow would be when self-spotting:
  - find suitable freq via CAT page VFO
  - call out to ensure it's open
  - <atu> button
Or, when pouncing:
  - get on freq via click-to-pounce
  - switch to CAT page
  - use VFO +/- 5 and hear nothing
  - <atu> button
  - reverse prior VFO button to go back to orig freq

Either way, the "ATU" button is all that's now necessary as I see it.

(P.S. yes #63 is still an issue, but separate from pure ATU.  Will tackle that differently)

Let me know if this doesn't match expectations.

poynting, 2025-05-23:
Yeah, that sounds good.

jeffkowalski, 2025-05-29:
ATU is now implemented in https://github.com/SOTAmat/SOTAcat/commit/12778b30191a8ee5132f0096b962aa255ff79b63

That closes this mega-issue.  We'll track additional feature requests in separate issues.