SOTAmat/SOTAcat #13: Implement OTA updates

jeffkowalski, 2024-02-22:
Brian says:
OTA updates would be wonderful.  However, we are nearly at the 2MB limit and OTA requires 3X the storage.  We can double our memory because PlatformIO thinks our board only has 2MB of storage when in fact it has 4MB.  We need to twiddle a setting.

jeffkowalski, 2024-03-24:
Just some background: https://blog.espressif.com/ota-updates-framework-ab5438e30c12

brianmathews, 2024-03-25:
Note, I found the bug where our module was being sensed as only 2MB Flash, when in fact we have 4MB flash.
That issue was fixed in an earlier check-in.
So OTA is very much possible now that we have access to the full 4MB.

jeffkowalski, 2024-03-25:
I think that was fixed in 923b60148b2fe55c2a9f7a633c6099469deb784d
I've been looking into OTA.
One thing to decide - does the user upload a .bin that they've downloaded from somewhere, or should we "check for updates" against a known URL?

brianmathews, 2024-03-25:
I would check for updates against a known location such as the sotamat.com web server. GitHub LFS does not allow us to purge old versions and storage counts against the free tier quota.  Hence we should use some non-GitHub web server.

We also want to make the process as simple as possible for non-developers. Developers can build their own source from GitHub.

jeffkowalski, 2024-04-17:
(Note to self, more background information) see https://gitlab.com/kevinwolfe/esp32_template/-/blob/87ca0ecd615115fdc33013d45b84a237da230575/main/main.c for example

jeffkowalski, 2024-04-18:
@brianmathews Please give me a URL on sotamat.com where I can check for "official" releases of SOTAcat.  I'd like to implement the "check for new version" feature, which is a component of OTA updates.

brianmathews, 2024-04-18:
You can name the file(s) anything you want.  They will just be part of the Wordpress Media folder as files in the pattern:

https://sotamat.com/wp-content/uploads/<YOUR-FILENAME-HERE>

jeffkowalski, 2024-04-18:
Excellent.
What will be the protocol for:
1. Identifying release-able commit-points
2. Building the code 
3. Uploading and naming the file.  (who has access)
4. Updating the "latest.txt" at the same place, if we assume to go that route.
Can you envision a script for 2-4?  MakeRelease.sh?  ;)

jeffkowalski, 2024-04-18:
> For the 1-click WebSerial deployment that is on the
> https://sotamat.com/sotacat page, the BIN file and version files are
> already scripted and part of the build process. For OTA we can extend the
> build process to make the required files on every build.
> When we want to publish I just drag-n-drop the set of files into
> Wordpress's Media Bin and voila.
> If you look at how the current unified BIN file is made in the
> platformio.ini file and the respective Python build extensions to
> PlatformIO you can see how you can do something very similar. for 2,3,4:
> it's just I'd do 2,3,4 on every build (if it is fast) and simply put those
> assets in the .gitignore. I can drag-n-drop into Wordpress when we want a
> real "release".

We may want a different unified BIN that doesn't blow away settings in
NVS.  Maybe one that just packs code and web assets.  Different build
script?

commit 81e9a782 (2024-08-23): fix #13: implement OTA updates

OTA firmware updates are now available through an interface at the
bottom of the Settings page.  This is available only on the SEED_XIAO
platform (which includes both AB6D_1 and K5EM_1 hardware versions),
simply because that platform has a corresponding partition table
defined.  It would be straightforward to extend support, via the
addition of an appropriate partition table.

This new implementation allows for uploading a file from client computer
to the SOTAcat.  It doesn't check for file version being valid, or more
recent, or anything, really.  It is therefo [... cut]
closed 2024-08-23 by jeffkowalski