SOTAmat/SOTAcat #15: Implement browser-based flashing

jeffkowalski, 2024-02-22:
Brian says:
Rather than, or in addition to, OTA updates, there are tools to allow installing firmware directly from a Chrome web browser (Chrome has WebUART ability!  Firefox does not).  This is a very simple way to allow field updates without all the storage requirements of OTA (which has to store the current and upgraded firmware).  While I love this Browser based flashing tool, I can’t seem to get it working even when I flash the different “pages” that PlatformIO builds.  https://espressif.github.io/esptool-js/  so we are close but not there.  Also, I would want a single unified “package” to flash the entire device, but the ESP Tool is set up to only flash the sub-packages individually.  Yuck.

See also #13

jeffkowalski, 2024-03-24:
Likely addressed by 22c777059f626083334e2ea604983732163dbded and prior
@brianmathews - done?

brianmathews, 2024-03-24:
Yes, there was some optional stuff I could do, but I think it would be another ticket.  Closing this one.

closed 2024-03-24 by brianmathews