SOTAmat/SOTAcat #6: Show a table for Alerts in addition to Spots

brianmathews, 2024-02-22:
Today SOTACAT displays a list of recent Spots for SOTA or POTA.

A feature request would be to see a list of upcoming Alerts (in particular SOTA Alerts) sorted by their time into the future...

jeffkowalski, 2025-05-23:
Given that there's no available action, nothing to "pounce upon" with those alerts, isn't it more reasonable to rely on other apps or, in fact the sotadata pages for that info?  What would the user really want to do?

jeffkowalski, 2026-02-19:
I'm going to close this.
There are other apps and websites that provide SOTA Alert data.
Viewing Alerts requires an internet access not always available to SOTAcat.
There is nothing to act upon in SOTAcat until there is an active spot.
This seems to be out of scope for SOTAcat UI.

closed 2026-02-19 by jeffkowalski