SOTAmat/SOTAcat #21: bug: split network not working on android

jeffkowalski, 2024-03-03:
Using a Google Pixel 7 running Android 14, the SOTAcat webpage does not load external data about sota and pota spots.
Connection via 192.168.4.1 in Chrome browser works just fine, and the webpage loads.  Controls from the webpage back to the phone work as designed. But the chart of spots never populates.

Brian reports that until he added a null gateway to wifi.cpp
```
    IP4_ADDR(&info.gw, 0, 0, 0, 0); // Zero gateway address
```
this didn't work on any android or ios device.

Perhaps there's something else that is malfunctioning now that we've moved to C++ and initialization of the ```wifi_config``` is incompletely specified in ```wifi_init_ap()```

jeffkowalski, 2024-03-03:
Ensuring all members of wifi config structures are initialized had no effect.  f7c6af4856837eaf5ad042db122e87d1f1cc3770 was not effective here; the symptoms are not improved.
On Android, if connected to the SOTAcat wifi AP, then access to internet is prevented to _any_ app, even though cellular internet is still active.

jeffkowalski, 2024-03-04:
Confirmed on an iphone today that all works as expected there, albeit strangely slow. Still not working on Android, as reported above.  Seems to be an Android-only issue (or maybe just my phone)?
Would appreciate more testers here.

poynting, 2024-03-05:
I see the same issue Jeff, running a Pixel 6 Pro also on Android 14.

jeffkowalski, 2024-03-05:
Thanks for the confirmation.  Good to know it's not just something wrong with my phone alone.
This may take some while to debug.  It's not obvious why ios and regular desktops are working OK, but Android is not.
I'm motivated, though - I use only Android phones.

poynting, 2024-03-06:
My current workaround is to enable my phone's hotspot and have SOTAcat connect to that. This may not be a great field solution as I'm not sure how stable the IP address will be, but for testing I can see the assigned address on the terminal monitor (and it's been stable for at least a session of testing).  This isn't a solution for users that don't have hotspot ability on their Android phone.

That does remind me that I need to sort out .local addressing. How are you handling discovery on Android Jeff?

jeffkowalski, 2024-03-07:
That's an awesome workaround.  In fact, I think we should support connecting to two client SSIDs by default.  I'm going to log an issue for that, so we connect preferentially to the presumed home SSID then to the phone hotspot.

Still, if you find the way to get split networking working on Android, please let me know.  Truthfully, I can't even discern the proper search terms online to see if others more generally have seen this problem.

poynting, 2024-03-07:
From my searching it's a common problem that I haven't seen any solutions to. I almost put in a feature request to support a primary and secondary host network, so I think we're on the same page with that being a good addition.

jeffkowalski, 2024-03-24:
Addressed via work-around: allow for two STA-mode wifi connections, one to cover hotspots suitable for Android.
Full preferences now implemented in https://github.com/SOTAmat/SOTAcat/commit/fb570e59f74094632f1362b9f68b0c3217aaf5b5

closed 2024-03-24 by jeffkowalski