SOTAmat/SOTAcat #24: Implement Unique WiFi SSIDs

poynting, 2024-03-05:
When multiple SOTAcats are powered on we can't currently tell which one we're connecting to over WiFi. At the moment this is more of a developer issue, but once they take over the world and we start having name collisions on summits it could become a user issue.

I suggest we have a factory/user settable serial number saved so each SOTAcat has a different default SSID, e.g. SOTAcat-1234. 

An extension of this could be making the broadcast SSID completely customizable, but we may still want some default option that includes a serial-number-like value added to a human-readable SSID.

jeffkowalski, 2024-03-05:
Yes, this was on my mental todo-list.  Thanks for taking the time to write it up.  We'll likely initially broadcast a "SOTAcatNNNN" where NNNN is some unique ID like macaddr, then have a settings page where one can adjust the SSID to their liking.
So this is really two feature requests:
1) to uniquely identify SOTAcat SSID at startup
2) to allow user-setting of SSID
I'll likely tackle the first one quickly, and work on the second a bit later.

jeffkowalski, 2024-03-07:
Fixed with poynting's (Justin McAllister's) commit.  Thanks, Justin!