SOTAmat/SOTAcat #25: Attempting to run direct frequency control FT8 results in mutex error

rexington, 2024-03-05:
Attempting to send an FT8 message with SOTAMat 1.2.1 (67) on iOS, the operation fails and the following is what shows up in the logs:
```
V (62084) sc:hdl_ft8.: request buffer[93] = "messageText=SM%20KE6MT%2F720F&timeNow=1709679838759&rfFrequency=14074000&audioFrequency=1553"
V (62104) sc:kx_cmds.: trace: get_kx_state()
V (62104) sc:kx_cmds.: trace: get_from_kx()
V (62104) sc:kx_cmds.: trace: uart_get_command()
E (62114) sc:kx_cmds.: RADIO PORT NOT LOCKED! (coding error in caller)
I (62134) sc:kx_cmds.: command 'FT;' returned 'FT0' after 10.871 ms
I (62134) sc:kx_cmds.: kx command 'FT' returns 0
V (62134) sc:kx_cmds.: trace: get_from_kx()
V (62134) sc:kx_cmds.: trace: uart_get_command()
E (62144) sc:kx_cmds.: RADIO PORT NOT LOCKED! (coding error in caller)
I (62174) sc:kx_cmds.: command 'FA;' returned 'FA00007235000' after 24.008 ms
I (62174) sc:kx_cmds.: kx command 'FA' returns 7235000
V (62174) sc:kx_cmds.: trace: get_from_kx_menu_item()
V (62184) sc:kx_cmds.: put_to_kx(MN) attempting value 58
V (62184) sc:kx_cmds.: trace: get_from_kx()
V (62194) sc:kx_cmds.: trace: uart_get_command()
E (62194) sc:kx_cmds.: RADIO PORT NOT LOCKED! (coding error in caller)
I (62234) sc:kx_cmds.: command 'MN;' returned 'MN058' after 27.811 ms
I (62234) sc:kx_cmds.: kx command 'MN' returns 58
I (62234) sc:kx_cmds.: command 'MN' successful; value = 58
V (62244) sc:kx_cmds.: trace: get_from_kx()
V (62244) sc:kx_cmds.: trace: uart_get_command()
E (62244) sc:kx_cmds.: RADIO PORT NOT LOCKED! (coding error in caller)
I (62264) sc:kx_cmds.: command 'MP;' returned 'MP000' after 12.300 ms
I (62264) sc:kx_cmds.: kx command 'MP' returns 0
V (62274) sc:kx_cmds.: put_to_kx(MN) attempting value 255
V (62274) sc:kx_cmds.: trace: get_from_kx()
V (62284) sc:kx_cmds.: trace: uart_get_command()
E (62284) sc:kx_cmds.: RADIO PORT NOT LOCKED! (coding error in caller)
I (62304) sc:kx_cmds.: command 'MN;' returned 'MN255' after 10.014 ms
I (62304) sc:kx_cmds.: kx command 'MN' returns 255
I (62304) sc:kx_cmds.: command 'MN' successful; value = 255
V (62314) sc:kx_cmds.: trace: get_from_kx()
V (62314) sc:kx_cmds.: trace: uart_get_command()
E (62324) sc:kx_cmds.: RADIO PORT NOT LOCKED! (coding error in caller)
I (62334) sc:kx_cmds.: command 'MD;' returned 'MD1' after 9.651 ms
I (62334) sc:kx_cmds.: kx command 'MD' returns 1
V (62334) sc:kx_cmds.: put_to_kx(MD) attempting value 3
V (62344) sc:kx_cmds.: trace: get_from_kx()
V (62344) sc:kx_cmds.: trace: uart_get_command()
E (62354) sc:kx_cmds.: RADIO PORT NOT LOCKED! (coding error in caller)
I (63084) sc:kx_cmds.: command 'MD;' returned 'MD3' after 719.129 ms
I (63084) sc:kx_cmds.: kx command 'MD' returns 3
I (63084) sc:kx_cmds.: command 'MD' successful; value = 3
V (63084) sc:kx_cmds.: trace: get_from_kx()
V (63094) sc:kx_cmds.: trace: uart_get_command()
E (63094) sc:kx_cmds.: RADIO PORT NOT LOCKED! (coding error in caller)
I (63114) sc:kx_cmds.: command 'AP;' returned 'AP0' after 8.975 ms
I (63114) sc:kx_cmds.: kx command 'AP' returns 0
V (63114) sc:kx_cmds.: put_to_kx(MD) attempting value 1
V (63124) sc:kx_cmds.: trace: get_from_kx()
V (63124) sc:kx_cmds.: trace: uart_get_command()
E (63134) sc:kx_cmds.: RADIO PORT NOT LOCKED! (coding error in caller)
V (63414) sc:hdl_batt: battery voltage: 4.095 V
V (63414) sc:idletask: heap: 289044 (used 90784, free 198260) [bytes]
V (63414) sc:idletask: blinks 3799289
I (63694) sc:kx_cmds.: command 'MD;' returned 'MD1' after 552.656 ms
I (63694) sc:kx_cmds.: kx command 'MD' returns 1
I (63694) sc:kx_cmds.: command 'MD' successful; value = 1
V (63694) sc:kx_cmds.: put_to_kx(FR) attempting value 0
V (63704) sc:kx_cmds.: trace: get_from_kx()
V (63704) sc:kx_cmds.: trace: uart_get_command()
E (63714) sc:kx_cmds.: RADIO PORT NOT LOCKED! (coding error in caller)
I (63784) sc:idletask: powering off due to inactivity
```

poynting, 2024-03-06:
I believe that this is fixed with [this PR](https://github.com/SOTAmat/SOTAcat/pull/26).

jeffkowalski, 2024-03-24:
Addressed with aforementioned PR and also subsequent commit da7779e607cba75be57eb13471ca50f762bf73e0

closed 2024-03-24 by jeffkowalski