SOTAmat/SOTAcat #32: create option for disable deep-sleep

jeffkowalski, 2024-03-07:
Perhaps deep-sleep is not useful when connected to certain SSIDs, like the home network, where it may be presumed the SOTAcat is plugged in to charge.

We offer this in advance of truly detecting charging state via hardware input.
Meanwhile, it's useful for debugging.

poynting, 2024-03-07:
I think staying alive when connected to certain APs is a good option. 

I also have USB charge detection working. I just put a voltage divider on the 5V rail and use GPIO3 to read it. In the idle task, if the GPIO is high, I keep blinks at 1 (but we could just as easily reset the user action time).  I'll certainly be adding this USB detect feature to my next revision hardware, so I think it's good to consider both as complimentary approaches.

jeffkowalski, 2024-03-07:
Excellent.  I'm eager to get my hands on your next-gen implementation, even if it's a breadboard schematic.  It's the future.
And - off topic - we should find a time to hike and wrestle this stuff.

jeffkowalski, 2024-03-24:
The disabling is implemented in a different, but equally effective way in 291d80bfc8abe82192830115ed9175bd09b1c3ff: when battery > 80% (i.e. charged and still plugged in).

closed 2024-03-24 by jeffkowalski