SOTAmat/SOTAcat #36: improve Settings UI

jeffkowalski, 2024-03-24:
Most important is danger of the hitting factory reset button without secondary confirmation, clearing wifi credentials.

Beyond that, I'd say the whole Settings page is ugly and could use a bunch of work to make it 
1) more aesthetic, and 
2) implement a variety of checks like 
    a) are you sure you want to factory reset?  and 
    b) checks to ensure ssids are either empty or >= 2 and  <= MAX_WIFI_SSID_SIZE chars and passwords are either empty or >= 8 and  <= MAX_WIFI_PASS_SIZE chars.  There are some rudiments of the latter already in the HTML, but no validation.

brianmathews, 2024-03-24:
I validated that the Web-Serial firmware flash code on https://sotamat.com/sotacat does overwrite the NVS settings and performs a full "factory reset".  Since I don't see a use case for a factory reset while in the field, I suggest we remove the factory reset button.  For people in-the-shack, they can use the firmware flash button with a USB cable which avoids accidents.
Agreed?

jeffkowalski, 2024-03-24:
Sure that's a reasonable way of erasing the NVS.
I'll take out the factory reset button and code.

brianmathews, 2024-03-24:
- Added more instructions on the settings page 

- Tried to improve the visual style.  Thank you ChatGPT.

- I removed the default strings and made them blank so that the code doesn't try to connect to an SSID when it has the default values for "<home network>" etc.  Instead the headers have the instructions for each Client-1, Client-2, Server rather than the textboxes having the instructions.

- Disabled entering deep sleep mode if the battery is more than 70% full.  This will almost always be true when plugged in via USB.

I did not add the requested value checking.

commit 2b8bcd37 (2024-04-02): fix #36: friendly validation of settings

Wifi settings now have two levels of validation
First, using HTML5's validations we check for proper lengths of SSID and
Password.  Then, using custom code we ensure that if one is set, so is
the other.
closed 2024-04-02 by jeffkowalski