SOTAmat/SOTAcat #38: Bug: On iOS the spots "Time-range" selector does nothing and does not persist.

brianmathews, 2024-03-25:
Bug: On iOS the spots "Time-range" selector does nothing and does not persist.

Also, the formatting is not aesthetically pleasing.

Also, the drop-down has crazy options that I think should get purged: such as 10 seconds.  The list only refreshes every 1 minute.  I would remove the 10 second and 1 minute options and add a 12 hour option (for the purpose of finding S2S mistakes at the end of the day by reviewing history).

jeffkowalski, 2024-03-25:
It's not a time-range selector, it's a refresh rate.
Does that clarify?

jeffkowalski, 2024-03-25:
I totally misunderstood #22, which I'll re-open.
I'll close this one as a duplicate.
I'll implement the lookback duration rather than polling rate.

closed 2024-03-25 by jeffkowalski