SOTAmat/SOTAcat #40: SOTAMAT not matching SOTACAT frequency on launch

jeffkowalski, 2024-04-15:
Brian reports:
I'm seeing a bug in SOTAMAT when launched from SOTACAT.  It no longer is matching a SOTAMAT frequency / mode pair to the radio's current state.  Likely a problem in SOTAMAT but seems like a new issue.

jeffkowalski, 2024-04-15:
If I have selected a frequency from the SOTACAT spot list, say 14.225, 
then press the SOTAMAT button, 
when SOTAMAT launches, 
I see it request the frequency and mode from the SOTACAT (`handler_frequency_get()` and `handler_mode_get()`).  
Both return valid values, 
but there is no UI change in the SOTAMAT.

This behavior is true whether or not SOTAMAT is initially showing the same mode/band that is on the radio.
@brianmathews, I'll need the SOTAMAT code to investigate further.

brianmathews, 2024-04-20:
I found the bug in SOTAMAT.  The issue is that SOTACAT used to tell SOTAMAT that it was using "SSB", but now SOTACAT tells SOTAMAT it is using "USB" or "LSB" and thus SOTAMAT didn't correctly find a match.  I am closing this bug and will fix the bug in SOTAMAT.

closed 2024-04-20 by brianmathews