SOTAmat/SOTAcat #46: improve detection of current location, for accurate distances

jeffkowalski, 2024-07-15:
The present implementation does not yield accurate operating location, leading to very skewed distance calculations.

While on Pine Mountain, https://sotl.as/summits/W6/NC-331 in Northern CA, on 2024-07-09, ip-api.com returned that I was at (36.102, -115.1447), which is in Paradise, NV, just South of Las Vegas.

We use ip-api.com in the [current code](https://github.com/SOTAmat/SOTAcat/blob/2bb037c9763735518264503ea5b7fb47b6afcf2c/src/web/main.js#L248) to get the apparent location of the operator.  This does not use the GPS on the phone.  Regrettably the very straightforward use of ```navigator.geolocation.getCurrentPosition``` is available only under HTTPS, so we have to use a service that attempts to locate the operator by IP, presumably by cell-location.  But we're getting a bad result.

Choices are:
1) Enter the location manually, perhaps the current park/summit code, and look that up
2) Find a better service than ```http://ip-api.com/json/?fields=status,message,lat,lon``` that offers greater accuracy
3) Implement HTTPS so we can use the GPS-based location.  Coding this is simple.  There is a minor performance hit for encryption/decryption.  The greatest challenge is determining what to do about certificate generation and distribution.

brianmathews, 2024-08-22:
...and Jeff also suggested option #4:
  4. Have SOTACAT make a "deep link" call to SOTAMAT app and ask SOTAMAT for the GPS location.  SOTAMAT would then return to the URL provided to it by SOTACAT with extra URL parameters for the current Lat/Lon.   There is a security/privacy concern with this and SOTAMAT won't be able to trust the origin of the inbound deep-link.  One solution is to ask the user to confirm they want to share GPS location with a web page.
  
  ...and Brian has a similar idea #5:
    5. Since we only need geolocation when we have internet (for the click-to-pounce distance table), we could have the JavaScript make build a 1-pixel <iframe> and that <iframe> could make a call to the https://sotamat.com web server using HTTPS, receive JavaScript via HTTPS from that server, get the GPS coordinates using the HTTPS based page, and then have the <iframe> javascript communicate with the parent page (HTTP) Javascript using the PostMessage API.

jeffkowalski, 2025-05-23:
Not a bad idea. It is tantamount to simply returning a perturbed GPS lat/lon from SOTAmat. If SOTAmat returned such a perturbed pair, it would leave the rest of the SOTAcat code intact.  Forget the array passing.  Just let's have a "close enough" present position.

jeffkowalski, 2025-06-08:
Not a fix, but a workaround: https://github.com/SOTAmat/SOTAcat/commit/6060aed04af72349a4f50b2fbdcd3132875aee1d
You can now override your apparent location by entering GPS coordinates on the "Settings" tab.
Your GPS location is easily copied from the SOTAmat page (even without network access), or any other suitable page (if you have network access).

chrisdebian, 2025-10-06:
Ok, again with some help; i've tried to keep the paragraph numbers flowing, so they fit in with the above:

Additional Options for GPS Accuracy Improvement

**Option 9: GPS Averaging & Filtering**
Implement GPS coordinate averaging over time with outlier rejection to smooth out GPS jitter and improve accuracy for stationary activations.

Implementation:
    • Collect multiple GPS readings over 30-60 seconds 
    • Apply Kalman filtering or simple moving average to smooth coordinate data 
    • Reject outliers that are significantly different from the mean 
    • Only update displayed location when confidence threshold is met 

Pros:
    • No privacy concerns - works entirely on device 
    • Significantly improves accuracy for stationary use (ideal for SOTA/POTA activations) 
    • Better results when more satellites are visible 
    • Low complexity to implement 

Cons:
    • Requires 30-60 seconds to get stable reading 
    • May not help if satellite reception is fundamentally poor 

**Option 10: HDOP/Accuracy Quality Indicators**
Display GPS quality metrics and only use location data when accuracy is sufficient. Provide user feedback about GPS status.

Implementation:
    • Read HDOP (Horizontal Dilution of Precision) from GPS module 
    • Display satellite count and signal strength to user 
    • Only update location when accuracy threshold is met (e.g., HDOP < 2.0) 
    • Provide UI feedback such as: "Waiting for GPS fix... (3/4 satellites)" 
    • Visual indicator (green/yellow/red) for GPS quality 

Pros:
    • User understands why location might be inaccurate or unavailable 
    • Prevents using unreliable GPS data 
    • No privacy concerns 
    • Educational - helps users optimize GPS reception 
    • Can be combined with other options 

Cons:
    • Doesn't improve hardware accuracy, just filters bad data 
    • May result in "no location available" more often in poor conditions 
    • Requires UI updates to display quality indicators 

**Option 11: Cached Last Known Good Location**
Cache the last accurate GPS reading and reuse it until better data becomes available, with clear age indicators.
Implementation:
    • Store last GPS fix with timestamp and accuracy metric 
    • If current GPS quality is poor, display cached location with age indicator 
    • UI displays: "Location: [coordinates] (cached, 5 minutes old)" 
    • Add setting for maximum cache age (e.g., 15 minutes) 
    • Clear cache if device detects movement (if accelerometer available) 

Pros:
    • Always shows a reasonable location to the user 
    • Excellent for stationary activations (most SOTA/POTA use cases) 
    • No privacy concerns 
    • Simple to implement 
    • Reduces user frustration with GPS dropouts 

Cons:
    • May be stale if user moves between activations 
    • Could mislead users if age indicator not prominently displayed 
    • Need clear UI to indicate cached vs. live GPS data 

**Recommended Implementation Approach**

These three options work well together and could be implemented in phases:

Phase 1: Option 10 (Quality Indicators)
    • Gives immediate user feedback 
    • Sets foundation for understanding GPS quality 

Phase 2: Option 9 (Averaging/Filtering)
    • Improves actual accuracy 
    • Works with quality indicators to show improvement 

Phase 3: Option 11 (Caching)
    • Fills gaps when GPS temporarily unavailable 
    • Provides best user experience for stationary activations 

All three options are device-local with no privacy implications and align well with the existing workaround (manual GPS override in Settings).

jeffkowalski, 2026-02-03:
The QRX page now uses a redirect through github pages to gain HTTPS allowing the mobile browser to permit precise GPS retrieval.

The main commit was d4c58c5 by Jeff Kowalski on Mon Dec 22 19:53:59 2025:
add HTTPS geolocation bridge for GitHub Pages
Bridge page requests browser location (requires HTTPS), then redirects back to
SOTAcat with coordinates.
  added two files:
  - docs/geolocation/bridge.js (180 lines)
  - docs/geolocation/index.html (123 lines)

The follow-up commit a545aee then wired it into the app ("add browser
geolocation via HTTPS bridge"), and 64e53bd later simplified the location UI by
removing the old IP-based geolocation in favor of this approach.