SOTAmat/SOTAcat #48: log shows incorrect wifi connection

jeffkowalski, 2024-08-14:
During setup, we try two clients in order, then default to access point mode.
Regrettably, this isn't getting logged correctly.
For example, with Client 1 = Pixel7 and Client2 = cvcc, and Pixel7 unavailable but cvcc operating, we see that the device correctly connects to Client2 (cvcc), but the logs are incorrect, showing that it was connected to Client1 (Pixel7)
```
I (2294) sc:wifi....: Attempting to connect to WiFi: Pixel7
I (2294) sc:wifi....: Default max TX power: 80
I (2294) sc:wifi....: Setting WiFi max power to 44
I (2294) sc:wifi....: Confirmed new max TX power: 44
I (2294) sc:wifi....: WiFi initialization complete. Waiting for connection...
I (5824) esp_netif_handlers: sta ip: 192.168.7.240, mask: 255.255.255.0, gw: 192.168.7.1
I (5824) sc:wifi....: Connected to network: Pixel7, IP address: 192.168.7.240
I (5824) sc:wifi....: Stopping AP mode
I (5824) wifi:mode : sta (d4:f9:8d:00:a4:80)
I (6194) sc:hdl_stat: returning connection status: 🟢
I (6254) sc:hdl_stat: returning connection status: 🟢
I (6294) sc:wifi....: WiFi connection established to Pixel7
```

jeffkowalski, 2024-08-14:
Here's a more complete log.  Again, note that while we connected to client2, it says we're connected to client1.
```
I (5214) sc:setup...: Starting WiFi task...
V (5214) sc:wifi....: trace: wifi_init()
I (5224) wifi:wifi driver task: 3fca4ad0, prio:23, stack:6656, core=0
I (5224) wifi:wifi firmware version: 91b9630
I (5224) wifi:wifi certification version: v7.0
I (5224) wifi:config NVS flash: enabled
I (5224) wifi:config nano formating: disabled
I (5224) wifi:Init data frame dynamic rx buffer num: 32
I (5224) wifi:Init static rx mgmt buffer num: 5
I (5224) wifi:Init management short buffer num: 32
I (5224) wifi:Init dynamic tx buffer num: 32
I (5224) wifi:Init static tx FG buffer num: 2
I (5224) wifi:Init static rx buffer size: 1600
I (5224) wifi:Init static rx buffer num: 10
I (5224) wifi:Init dynamic rx buffer num: 32
I (5234) wifi_init: rx ba win: 6
I (5234) wifi_init: tcpip mbox: 32
I (5234) wifi_init: udp mbox: 6
I (5234) wifi_init: tcp mbox: 6
I (5234) wifi_init: tcp tx win: 5744
I (5234) wifi_init: tcp rx win: 5744
I (5234) wifi_init: tcp mss: 1440
I (5234) wifi_init: WiFi IRAM OP enabled
I (5234) wifi_init: WiFi RX IRAM OP enabled
V (5244) sc:wifi....: trace: wifi_init_ap()
I (5244) sc:wifi....: Initializing WiFi AP with SSID: SOTAcat-2E38
I (5244) sc:wifi....: Configuring DHCP server for AP mode.
I (5244) sc:wifi....: Current WiFi mode: 3
I (5244) sc:wifi....: WiFi AP setup complete. SSID: SOTAcat-2E38, Channel: 1
I (5274) wifi:mode : sta (ec:da:3b:36:2e:38) + softAP (ec:da:3b:36:2e:39)
I (5274) wifi:enable tsf
I (5274) wifi:Total power save buffer number: 16
I (5274) wifi:Init max length of beacon: 752/752
I (5274) wifi:Init max length of beacon: 752/752
I (5284) sc:wifi....: STA mode started. Attempting initial connection to SSID1: Pixel7
I (5284) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (5284) wifi:ap channel adjust o:1,1 n:6,2
I (5284) wifi:new:<6,0>, old:<1,1>, ap:<6,2>, sta:<6,0>, prof:1
I (5284) wifi:state: init -> auth (b0)
I (5324) wifi:state: auth -> assoc (0)
I (5334) wifi:state: assoc -> run (10)
I (5364) wifi:connected with cvcc, aid = 16, channel 6, BW20, bssid = 60:5f:8d:74:8e:06
I (5364) wifi:security: WPA2-PSK, phy: bgn, rssi: -71
I (5364) wifi:pm start, type: 1
I (5364) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
I (5424) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (5874) wifi:<ba-add>idx:0 (ifx:0, 60:5f:8d:74:8e:06), tid:0, ssn:0, winSize:64
V (7284) sc:wifi....: trace: wifi_init_sta(ssid = 'Pixel7')
I (7294) sc:wifi....: Attempting to connect to WiFi: Pixel7
V (7294) sc:wifi....: trace: wifi_attenuate_power()
I (7294) sc:wifi....: Default max TX power: 80
I (7294) sc:wifi....: Setting WiFi max power to 44
I (7294) sc:wifi....: Confirmed new max TX power: 44
I (7294) sc:wifi....: WiFi initialization complete. Waiting for connection...
I (7864) sc:wifi....: Connected to network: Pixel7, IP address: 192.168.7.200
I (7864) sc:wifi....: Stopping AP mode
I (7864) wifi:mode : sta (ec:da:3b:36:2e:38)
I (8294) sc:wifi....: WiFi connection established to Pixel7
I (8294) sc:setup...: wifi initialized.
```

jeffkowalski, 2024-08-14:
By adding some additional logging, (which I'll commit shortly), the timing has changed, and we get a different result.
It does seem, however, that we're getting a ```WIFI_EVENT_STA_START``` before we actually init station mode.

Here's the log:

```
I (5214) sc:setup...: Starting WiFi task...
V (5214) sc:wifi....: trace: start_wifi_task()
V (5214) sc:wifi....: trace: wifi_task()
V (5214) sc:wifi....: trace: wifi_init()
I (5224) wifi:wifi driver task: 3fca4ad0, prio:23, stack:6656, core=0
I (5224) wifi:wifi firmware version: 91b9630
I (5224) wifi:wifi certification version: v7.0
I (5224) wifi:config NVS flash: enabled
I (5224) wifi:config nano formating: disabled
I (5224) wifi:Init data frame dynamic rx buffer num: 32
I (5224) wifi:Init static rx mgmt buffer num: 5
I (5224) wifi:Init management short buffer num: 32
I (5224) wifi:Init dynamic tx buffer num: 32
I (5224) wifi:Init static tx FG buffer num: 2
I (5224) wifi:Init static rx buffer size: 1600
I (5224) wifi:Init static rx buffer num: 10
I (5224) wifi:Init dynamic rx buffer num: 32
I (5234) wifi_init: rx ba win: 6
I (5234) wifi_init: tcpip mbox: 32
I (5234) wifi_init: udp mbox: 6
I (5234) wifi_init: tcp mbox: 6
I (5234) wifi_init: tcp tx win: 5744
I (5234) wifi_init: tcp rx win: 5744
I (5234) wifi_init: tcp mss: 1440
I (5234) wifi_init: WiFi IRAM OP enabled
I (5234) wifi_init: WiFi RX IRAM OP enabled
V (5244) sc:wifi....: trace: wifi_init_ap()
I (5244) sc:wifi....: Initializing WiFi AP with SSID: SOTAcat-2E38
I (5244) sc:wifi....: Configuring DHCP server for AP mode.
I (5244) sc:wifi....: Current WiFi mode: 3
I (5244) sc:wifi....: WiFi AP setup complete. SSID: SOTAcat-2E38, Channel: 1
I (5274) wifi:mode : sta (ec:da:3b:36:2e:38) + softAP (ec:da:3b:36:2e:39)
I (5274) wifi:enable tsf
I (5274) wifi:Total power save buffer number: 16
I (5284) wifi:Init max length of beacon: 752/752
I (5284) wifi:Init max length of beacon: 752/752
V (5284) sc:wifi....: trace: wifi_event_handler(event_base = 'WIFI_EVENT', event_id = 2)
I (5284) sc:wifi....: STA mode started. Attempting initial connection to SSID1: Pixel7
I (5284) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
V (5284) sc:wifi....: trace: wifi_event_handler(event_base = 'WIFI_EVENT', event_id = 12)
V (7284) sc:wifi....: trace: wifi_init_sta(ssid = 'Pixel7')
E (7284) wifi:sta is connecting, return error
E (7284) sc:wifi....: Failed to connect to WiFi: ESP_ERR_WIFI_CONN
V (7284) sc:wifi....: trace: wifi_attenuate_power()
I (7284) sc:wifi....: Default max TX power: 80
I (7284) sc:wifi....: Setting WiFi max power to 44
I (7284) sc:wifi....: Confirmed new max TX power: 44
I (7284) sc:wifi....: WiFi initialization complete. Waiting for connection...
V (8114) sc:wifi....: trace: wifi_event_handler(event_base = 'WIFI_EVENT', event_id = 5)
I (8114) sc:wifi....: Initial connection attempt to Pixel7 failed.
I (8114) sc:wifi....: Quick reconnect attempt 1/1 to Pixel7
V (10944) sc:wifi....: trace: wifi_event_handler(event_base = 'WIFI_EVENT', event_id = 5)
I (10944) sc:wifi....: Attempting to connect to SSID2: cvcc
V (10944) sc:wifi....: trace: wifi_init_sta(ssid = 'cvcc')
I (10954) sc:wifi....: Attempting to connect to WiFi: cvcc
I (10964) wifi:ap channel adjust o:1,1 n:6,1
I (10964) wifi:new:<6,1>, old:<1,1>, ap:<6,1>, sta:<6,1>, prof:1
I (10964) wifi:state: init -> auth (b0)
I (10964) wifi:state: auth -> assoc (0)
I (10974) wifi:state: assoc -> run (10)
I (10994) wifi:connected with cvcc, aid = 12, channel 6, 40U, bssid = f8:bb:bf:30:db:c6
I (10994) wifi:security: WPA2-PSK, phy: bgn, rssi: -45
I (11004) wifi:pm start, type: 1
I (11004) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
V (11004) sc:wifi....: trace: wifi_event_handler(event_base = 'WIFI_EVENT', event_id = 4)
I (11024) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (12164) wifi:<ba-add>idx:0 (ifx:0, f8:bb:bf:30:db:c6), tid:0, ssn:1, winSize:64
V (13004) sc:wifi....: trace: wifi_event_handler(event_base = 'IP_EVENT', event_id = 0)
I (13004) sc:wifi....: Connected to network: cvcc, IP address: 192.168.7.200
V (13004) sc:wifi....: trace: stop_ap_mode()
I (13004) sc:wifi....: Stopping AP mode
I (13004) wifi:mode : sta (ec:da:3b:36:2e:38)
V (13054) sc:wifi....: trace: wifi_event_handler(event_base = 'WIFI_EVENT', event_id = 13)
I (13284) sc:wifi....: WiFi connection established to cvcc
I (13284) sc:setup...: wifi initialized.
```

jeffkowalski, 2024-08-14:
Very strange to see ```I (5274) wifi:mode : sta (ec:da:3b:36:2e:38) + softAP (ec:da:3b:36:2e:39)``` in that second log.  Somehow, we've got two modes going on when we expect only STA mode.  Could this be the reason we get an unexpectedly early ```WIFI_EVENT_STA_START```?

jeffkowalski, 2024-08-18:
I've confirmed the problem exists, even when we never ask for a STA mode connection!
It looks like an issue with the way the AP is getting set up.
I added an infinite delay just after the call to ```wifi_init_ap()```:
```
    // Initialize AP mode
    wifi_init_ap();

    // Start WiFi and give the Access Point time to start
    ESP_ERROR_CHECK (esp_wifi_start());
    while (true)
        vTaskDelay (pdMS_TO_TICKS (2000));

    // Now initialize STA mode
    wifi_init_sta (g_sta1_ssid, g_sta1_pass);
```
Note that we'll never get to that ```wifi_init_sta call```
Nonetheless, we get ```WIFI_EVENT_STA_START``` event.  See below for the line I've highlighted with "**".  We're getting an event for a STA mode, but we haven't asked for STA mode.  This sets off a cascade of trouble.
```
I (5251) sc:setup...: Starting WiFi task...
V (5251) sc:wifi....: trace: start_wifi_task()
V (5251) sc:wifi....: trace: wifi_task()
V (5251) sc:wifi....: trace: wifi_init()
I (5251) pp: pp rom version: 9387209
I (5251) net80211: net80211 rom version: 9387209
I (5251) sc:setup...: Starting radio connection task...
I (5251) sc:setup...: Attempting to connect to radio...
D (5251) sc:lockable: locking radio
D (5251) sc:lockable: radio LOCKED --
V (5251) sc:kx_radio: trace: connect()
I (5261) wifi:wifi driver task: 3fca6240, prio:23, stack:6656, core=0
I (5261) wifi:wifi firmware version: a9f5b59
I (5261) wifi:wifi certification version: v7.0
I (5261) wifi:config NVS flash: enabled
I (5261) wifi:config nano formating: disabled
I (5261) wifi:Init data frame dynamic rx buffer num: 32
I (5261) wifi:Init static rx mgmt buffer num: 5
I (5261) wifi:Init management short buffer num: 32
I (5261) wifi:Init dynamic tx buffer num: 32
I (5261) wifi:Init static tx FG buffer num: 2
I (5261) wifi:Init static rx buffer size: 1600
I (5261) wifi:Init static rx buffer num: 10
I (5261) wifi:Init dynamic rx buffer num: 32
I (5261) wifi_init: rx ba win: 6
I (5271) wifi_init: tcpip mbox: 32
I (5271) wifi_init: udp mbox: 6
I (5271) wifi_init: tcp mbox: 6
I (5271) wifi_init: tcp tx win: 5744
I (5271) wifi_init: tcp rx win: 5744
I (5271) wifi_init: tcp mss: 1440
I (5271) wifi_init: WiFi IRAM OP enabled
I (5271) wifi_init: WiFi RX IRAM OP enabled
V (5281) sc:wifi....: trace: wifi_init_ap()
I (5281) sc:wifi....: Initializing WiFi AP with SSID: SOTAcat-2E38
I (5281) sc:wifi....: Configuring DHCP server for AP mode.
I (5281) sc:wifi....: Current WiFi mode: 3
I (5281) sc:wifi....: WiFi AP setup complete. SSID: SOTAcat-2E38, Channel: 1
I (5281) phy_init: phy_version 1150,7c3c08f,Jan 24 2024,17:32:21
I (5311) wifi:mode : sta (ec:da:3b:36:2e:38) + softAP (ec:da:3b:36:2e:39)
I (5311) wifi:enable tsf
I (5311) wifi:Total power save buffer number: 16
I (5321) wifi:Init max length of beacon: 752/752
I (5321) wifi:Init max length of beacon: 752/752
V (5321) sc:wifi....: trace: wifi_event_handler(event_base = 'WIFI_EVENT', event_id = 40)
V (5321) sc:wifi....: trace: wifi_event_handler(event_base = 'WIFI_EVENT', event_id = 2)
**I (5321) sc:wifi....: STA mode started. Attempting initial connection to SSID1: Pixel7**
I (5321) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (5321) wifi:ap channel adjust o:1,1 n:6,2
I (5321) wifi:new:<6,2>, old:<1,1>, ap:<6,2>, sta:<6,2>, prof:1
I (5321) wifi:state: init -> auth (b0)
V (5331) sc:wifi....: trace: wifi_event_handler(event_base = 'WIFI_EVENT', event_id = 12)
V (5331) sc:wifi....: trace: wifi_event_handler(event_base = 'WIFI_EVENT', event_id = 40)
I (5341) wifi:state: auth -> assoc (0)
I (5341) wifi:state: assoc -> run (10)
I (5361) wifi:connected with cvcc, aid = 40, channel 6, 40D, bssid = 60:5f:8d:74:8e:06
I (5361) wifi:security: WPA2-PSK, phy: bgn, rssi: -68
I (5361) wifi:pm start, type: 1
I (5361) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (5361) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
V (5361) sc:wifi....: trace: wifi_event_handler(event_base = 'WIFI_EVENT', event_id = 4)
I (5441) wifi:dp: 2, bi: 102400, li: 4, scale listen interval from 307200 us to 409600 us
I (5441) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (5751) sc:kx_radio: no response received for baud rate 38400
I (5871) wifi:<ba-add>idx:0 (ifx:0, 60:5f:8d:74:8e:06), tid:0, ssn:0, winSize:64
I (6251) sc:kx_radio: no response received for baud rate 19200
I (6751) sc:kx_radio: no response received for baud rate 9600
I (6861) esp_netif_handlers: sta ip: 192.168.7.200, mask: 255.255.255.0, gw: 192.168.7.1
V (6861) sc:wifi....: trace: wifi_event_handler(event_base = 'IP_EVENT', event_id = 0)
I (6861) sc:wifi....: Connected to network: Pixel7, IP address: 192.168.7.200
V (6861) sc:wifi....: trace: stop_ap_mode()
I (6861) sc:wifi....: Stopping AP mode
I (6861) wifi:mode : sta (ec:da:3b:36:2e:38)
V (6871) sc:wifi....: trace: wifi_event_handler(event_base = 'WIFI_EVENT', event_id = 13)
```

jeffkowalski, 2024-08-19:
this seems suspect:
```
ESP_ERROR_CHECK (esp_wifi_set_mode (WIFI_MODE_APSTA));
```
should it not be:
```
ESP_ERROR_CHECK (esp_wifi_set_mode (WIFI_MODE_AP));
```
Why would we want STA mode when we're setting up the AP?
See https://github.com/SOTAmat/SOTAcat/blob/5a4575a4ab6d5328856fdef895ddee9b7862f7cd/src/wifi.cpp#L263 and https://github.com/SOTAmat/SOTAcat/blob/5a4575a4ab6d5328856fdef895ddee9b7862f7cd/src/wifi.cpp#L196

If I make the change to ```WIFI_MODE_AP```, I no longer see the spurious event.  However, the wifi_setup declares success after setting up the AP and doesn't allow subsequent setup of STA.

Seems to me this approach of setting up AP mode first is broken.  AP mode should be the last resort after both STA ssids fail.

brianmathews, 2024-08-19:
We want to be in BOTH Station mode and Access Point mode simultaneously until we get a connection.  That's what WIFI_MODE_APSTA is supposed to do.
 
If we are AP mode only, then the CAT won't be able to connect to cell-phone Hotspots (ex. Android), or Home WiFi.
If we are STA mode only, then cell-phones (ex. iOS) that want to connect to the CAT won't have something to connect to.

The problem with the old version (where we start out trying to connect to Hotspots/HomeWiFi for a few seconds, and if that fails drop into AP mode) is that if someone first turns on the SOTACAT and then turns on their Android hotspot, the SOTACAT will have already given up on trying the STA mode and it will fail.  This has confused users who had failed attempts because of the order they did things in.

With WIFI_MODE_APSTA we are both an AP and a STA at the same time.  The STA thread will continuously attempt to connect to a Hotspot or Home-Wifi, while the AP allows iOS devices to connect to the AP.  As soon as one is connected to, the other mode is shut down.

I think I see the problem described in this thread:  [https://esp32.com/viewtopic.php?t=538](https://esp32.com/viewtopic.php?t=538)

I don't think my code is treating the APSTA mode as two different interfaces.  I think my code might be overloading the same interface for both modes.  The thread explains how to do it...

brianmathews, 2024-08-19:
This sample might help:
https://github.com/nopnop2002/esp-idf-wifi-apsta

jeffkowalski, 2024-08-19:
I was just looking at that - 
https://github.com/vonnieda/esp32-example-wifi-softap-and-sta/blob/main/main/main.c
https://github.com/nopnop2002/esp-idf-wifi-apsta/blob/main/main/main.c

The trick will be how to turn a single STA into a trial of two STAs for our prioritized STA SSIDs.  Note that in SoftAP mode, the AP and STA are configured together and the esp_wifi_connect is called once thereafter.

brianmathews, 2024-08-20:
Might be fixed in f0273dc
Please verify.

jeffkowalski, 2024-08-22:
I tried both hardware types, and the seem to be behaving better.  I can connect to SSID1 (Pixel7), or with that unavailable, watch the code connect to SSID2 (cvcc).  Access to the SOTAcat from my home wifi has not been as consistent as with the phone hotspot.  Occasionally I cannot find the dongle by name or ip address, even though I saw the wifi setup happen correctly in the logs.  This is hard to reproduce, however.

Overall, your recent commit seems to be a big improvement, Brian.  I would ask though to replace the structure inits with a memset(...0...), followed by setting of only those fields we mean to use non-default values.  The code is hard to read currently since it's hard to discern which fields are truly set to something special, versus a zero or #define'd constant that's effectively zero.

jeffkowalski, 2024-08-24:
I've run a test under most recent commit and am still seeing some issues in what I would presume to be at typical mode of running the sotacat at home (rather than in the field).
In my case, my SSID1 is my phone, ```Pixel7```, and SSID2 is my home, ```cvcc```.  Note in the log that ```Pixel7```fails because the hotspot is not turned on, and the SSID2 home connection to ```cvcc``` appears to work. What then is the error message ```E (12611) wifi:sta is connecting, return error``` (see line beginning with "E")?

```
I (6611)  sc:wifi....: attempting ssid: Pixel7
I (6611)  sc:wifi....: connecting to ap ssid: Pixel7
I (9441)  sc:wifi....: received event WIFI_EVENT_STA_DISCONNECTED
I (12271) sc:wifi....: received event WIFI_EVENT_STA_DISCONNECTED
I (12611) sc:wifi....: attempting ssid: cvcc
I (12611) sc:wifi....: connecting to ap ssid: cvcc
E (12611) wifi:sta is connecting, return error
I (13071) wifi:ap channel adjust o:1,1 n:6,1
I (13071) wifi:new:<6,1>, old:<1,1>, ap:<6,1>, sta:<6,1>, prof:1
I (13071) wifi:state: init -> auth (b0)
I (14071) wifi:state: auth -> init (200)
I (14071) wifi:new:<6,0>, old:<6,1>, ap:<6,1>, sta:<6,1>, prof:1
I (14071) sc:wifi....: received event WIFI_EVENT_STA_DISCONNECTED
I (14081) wifi:new:<6,1>, old:<6,0>, ap:<6,1>, sta:<6,1>, prof:1
I (14081) wifi:state: init -> auth (b0)
I (14091) wifi:state: auth -> assoc (0)
I (14091) wifi:state: assoc -> run (10)
I (14121) wifi:connected with cvcc, aid = 20, channel 6, 40U, bssid = f8:bb:bf:30:db:c6
I (14121) wifi:security: WPA2-PSK, phy: bgn, rssi: -49
I (14121) wifi:pm start, type: 1
I (14121) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (14121) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
I (14121) sc:wifi....: received event WIFI_EVENT_STA_CONNECTED, recording connected
I (14131) wifi:dp: 2, bi: 102400, li: 4, scale listen interval from 307200 us to 409600 us
I (14131) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (14211) sc:wifi....: starting mdns service
I (14221) sc:wifi....: mdns service started
I (14221) sc:setup...: wifi initialized.
I (14221) sc:wifi....: starting mdns service
I (14221) sc:wifi....: mdns service started
I (14221) sc:setup...: mdns initialized.
I (14221) sc:webserve: defined webserver callbacks.
I (14221) sc:setup...: webserver initialized.
```

brianmathews, 2024-08-25:
OK, this time I think it is fairly solid in several respects.  

See d68884d and c21016f  and ef54286 for all the changes.

I tested with:
- Android hotspot as station connecting to SSID-1, 
- iPhone split networking as an Access Point
- Home network with SSID-2
- No network present and have it cycle between attempts on SSID-1 and SSID-2, all while advertising an AP Access Point at the same time.
- mDNS seems to be working solidly
- DHCP seems to be working solidly

I think the only case that might have issues is when there is a disconnection.  It handles disconnects some times, but I think I saw a failure.  The workaround is to just power-cycle.  I think it is intermittent, or only in certain sequences, or it might not be a bug at all.  For now the main cases seem to work for me in a much more solid and predictable way even in Release mode.

After some testing I am hoping we can close this one!

brianmathews, 2024-08-27:
While there is still a WiFi failure mode, it seems to be a different edge case that happens after some disconnection events.  I haven't found a reliable sequence to reproduce it, but I am going to close this ticket as done and if we find the other bug we can open a new ticket for that issue.

closed 2024-08-27 by brianmathews