SOTAmat/SOTAcat #51: Radio not connected after firmware update

benmarwick, 2024-08-15:
I just got a SOTAcat from @poynting today, and have been enjoying controlling my KX3 with my Pixel 7. What an efficient way to operate!

I just now updated the firmware following the instructions here https://sotamat.com/sotacat/. After that, my device doesn't seem to be able to connect to the KX3. I no longer see the orange LED light up at all. When I power on, the blue LED flashes a few times, then is solid. I can access the device on 10.0.0.138 on my phone through my home wifi and it reports the battery details, but shows "??" where the radio connection indicator is. I have rebooted it several times from the page, the blue LED flashed a few times and I get a dialog box saying 'settings saved successfully... is now rebooting'. But no connection to the radio. 

Here's some of the logs, let me know if you need to see more: 

I (134329) sc:kx_radio: no response received for baud rate 4800
I (134829) sc:kx_radio: no response received for baud rate 38400
E (134869) sc:webserve: radio not connected
W (134869) httpd_txrx: httpd_resp_send_err: 500 Internal Server Error - Server has encountered an unexpected error
W (134869) httpd_uri: httpd_uri: uri handler execution failed
I (135329) sc:kx_radio: no response received for baud rate 19200
I (135829) sc:kx_radio: no response received for baud rate 9600
I (136329) sc:kx_radio: no response received for baud rate 4800
I (136829) sc:kx_radio: no response received for baud rate 38400
I (137329) sc:kx_radio: no response received for baud rate 19200
I (137829) sc:kx_radio: no response received for baud rate 9600
I (138329) sc:kx_radio: no response received for baud rate 4800
I (138829) sc:kx_radio: no response received for baud rate 38400

jeffkowalski, 2024-08-15:
Is there a version number in the web interface at upper left where you saw the "??" for radio connection?
I'm thinking that you may be running a pre-built release from the SOTAMAT website, which is far out of date, and was released prior to the release of K5EM's hardware which communicates with the radio very differently (literally inverted, or uninverted...nevermind).  If this is the case, the best cure is to build from the head of this repo.

brianmathews, 2024-08-15:
The K5EM version needs different firmware because he changed the circuit from my version.  The pinouts differ.  We are working on adding sensing logic so that we can have universal firmware.  Jeff KC6X just checked in changes to support that but until I get home you’ll have to build from source or wait until this weekend.
    -Brian Mathews

 #51)

benmarwick, 2024-08-15:
Thanks for your quick replies! I had a chat with Justin also, and following the tips from him and you all I built from source and uploaded to the device, and I'm back in action. So that's fantastic, I'm delighted. Thank you all for your hard work on this wonderful accessory for the KX3. My current version is 240815:1524-D

closed 2024-08-15 by benmarwick