SOTAmat/SOTAcat #56: split networking fails on android  (recent)

jeffkowalski, 2024-09-26:
in the latest build, split networking fails on android

steps to reproduce:
0. confirm wifi available and mobile (e.g. LTE) active connections on phone
1. offer hotspot from phone
2. connection from sotacat to phone's hotspot wifi
3. open browswer, using Bonjour browser's discovery of SOTAcat on phone

Notice that the web page populates fine, based on pages delivered from SOTAcat to phone.
However, no SOTA nor POTA data can be remotely retrieved, because the phone sees the SOTAcat as the internet gateway (incorrect).

If the internet access were split from the phone, routed to the SOTAcat for its traffic, and via mobile for the rest, then the spotting pages would be populated.

This was working recently.  A more recent commit appears to have broken the function.
That's a drag, since it eliminates "pounce" capability in the field.

jeffkowalski, 2024-09-28:
I'm thinking it's this that you removed:
https://github.com/SOTAmat/SOTAcat/commit/a121dac2f20492621b4641ce629e5ab2c73b4e0b#diff-cc284e969905e4c8b811640cd5de74cbffa38e1674797e6b3f26383d2c85f033L398-L409

Even though it appears to be AP-centric, it seems to me that it executes whenever the AP is active, and so applies to the Client as well.  The client case is what is broken for me on android.

brianmathews, 2024-09-30:
Fixed regression bug in 30bc0b4 (I hope).  I did a quick test and it seems to work on Android now.  I'm not sure how this passed my testing on iOS.  I might have tested the wrong SOTACAT (I have many laying around)!
In my last "improved" networking code I accidentally removed the DHCP code that tells the phone not to route through the SOTACAT  (gateway 0.0.0.0).

closed 2024-09-30 by brianmathews