SOTAmat/SOTAcat #74: build broken

jeffkowalski, 2025-08-04:
Latest changes appear to omit a reference to smbus

```
~/D/w/motes/SOTAcat [main] λ cd ..

~/D/workspace/motes λ rm -rf SOTAcat

~/D/workspace/motes λ git clone git@github.com:SOTAmat/SOTAcat.git
Cloning into 'SOTAcat'...
X11 forwarding request failed on channel 0
remote: Enumerating objects: 2014, done.
remote: Counting objects: 100% (445/445), done.
remote: Compressing objects: 100% (134/134), done.
remote: Total 2014 (delta 329), reused 367 (delta 272), pack-reused 1569 (from 1)
Receiving objects: 100% (2014/2014), 7.20 MiB | 1.89 MiB/s, done.
Resolving deltas: 100% (1354/1354), done.

~/D/workspace/motes λ cd SOTAcat

~/D/w/motes/SOTAcat [main] λ platformio run
Processing seeed_xiao_esp32c3_debug (board: seeed_xiao_esp32c3; platform: espressif32 @ 6.6.0; framework: espidf)
--------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
SOTACAT Pre-build step: Build type set to Debug
SOTACAT Pre-build step: Updated version in firmware/webtools/manifest.json to 2025-08-03_19:35-Debug
SOTACAT Pre-build step: Updated include/build_info.h with build date/time 250803:1935
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/seeed_xiao_esp32c3.html
PLATFORM: Espressif 32 (6.6.0) > Seeed Studio XIAO ESP32C3
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.50201.240515 (5.2.1) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-riscv32-esp @ 13.2.0+20230928
Reading CMake configuration...
Generating assembly for certificate bundle...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 3 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ft8_encoder
Building in release mode
Generating assembly for .pio/build/seeed_xiao_esp32c3_debug/index.html.S
Generating assembly for .pio/build/seeed_xiao_esp32c3_debug/style.css.S
Generating assembly for .pio/build/seeed_xiao_esp32c3_debug/main.js.S
Generating assembly for .pio/build/seeed_xiao_esp32c3_debug/sclogo.png.S
Generating assembly for .pio/build/seeed_xiao_esp32c3_debug/favicon.ico.S
Generating assembly for .pio/build/seeed_xiao_esp32c3_debug/sota.html.S
Generating assembly for .pio/build/seeed_xiao_esp32c3_debug/sota.js.S
Generating assembly for .pio/build/seeed_xiao_esp32c3_debug/pota.html.S
Generating assembly for .pio/build/seeed_xiao_esp32c3_debug/pota.js.S
Generating assembly for .pio/build/seeed_xiao_esp32c3_debug/settings.html.S
Generating assembly for .pio/build/seeed_xiao_esp32c3_debug/settings.js.S
Generating assembly for .pio/build/seeed_xiao_esp32c3_debug/cat.html.S
Generating assembly for .pio/build/seeed_xiao_esp32c3_debug/cat.js.S
Generating assembly for .pio/build/seeed_xiao_esp32c3_debug/about.html.S
Compiling .pio/build/seeed_xiao_esp32c3_debug/src/battery_monitor.o
Compiling .pio/build/seeed_xiao_esp32c3_debug/src/enter_deep_sleep.o
Compiling .pio/build/seeed_xiao_esp32c3_debug/src/handler_atu.o
Compiling .pio/build/seeed_xiao_esp32c3_debug/src/handler_battery.o
In file included from src/battery_monitor.cpp:4:
include/max17260.h:3:10: fatal error: smbus.h: No such file or directory

***************************************************************
* Looking for smbus.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:smbus.h"
* Web  > https://registry.platformio.org/search?q=header:smbus.h
*
***************************************************************

    3 | #include "smbus.h"
      |          ^~~~~~~~~
compilation terminated.
Compiling .pio/build/seeed_xiao_esp32c3_debug/src/handler_cat.o
*** [.pio/build/seeed_xiao_esp32c3_debug/src/battery_monitor.o] Error 1
========================================================== [FAILED] Took 13.30 seconds ==========================================================

Environment               Status    Duration
------------------------  --------  ------------
seeed_xiao_esp32c3_debug  FAILED    00:00:13.295
===================================================== 1 failed, 0 succeeded in 00:00:13.295 =====================================================

```

brianmathews, 2025-08-04:
@jeffkowalski  I think you didn't deal with the git submosules.  SMBUS is code coming from a different git repository as a git submodule.

The readme had instructions about how to clone this.  I just added some additional instructions if you neglected to follow the installation step.  Try this:

`git submodule update --init --recursive`

jeffkowalski, 2025-08-07:
no need for submodules
fixed using PlatformIO lib_deps in 82d567ab489816cae4710d6a54b75f1ea3cec7ca

closed 2025-08-07 by jeffkowalski